arduino/arduino-lint #158: Where can I find the list of rules?

ianfixes, 2021-01-12:
Is there a published list of all the linter rules? I couldn't find a reference to it in the documentation.

Alternately, is there a sample project that (on purpose) triggers every possible linter error?

per1234, 2021-01-12:
>Is there a published list of all the linter rules? 

Here are all the rules:
https://github.com/arduino/arduino-lint/blob/main/internal/rule/ruleconfiguration/ruleconfiguration.go#L60-L2859

You can get details on all the rules that are run for a given project type and tool configuration by running `arduino-lint --format json --verbose` or print it to a file using the `--report-file` flag.

>is there a sample project that (on purpose) triggers every possible linter error?

That would be one seriously messed up platform! It actually wouldn't be possible to trigger all of them from a single project because some are mutually exclusive.

There is an individual test project for each rule violation here:
https://github.com/arduino/arduino-lint/tree/main/internal/rule/rulefunction/testdata
It's pretty fun to run `arduino-lint --recursive` on that folder!

ianfixes, 2021-01-12:
OK, great.  I'll leave this issue open in case you want to convert it to a "publish the rules on a webpage" request, but my needs here are met.