arduino/arduino-lint #169: Please add check for invalid examples e.g. containing no ino file

ArminJo, 2021-04-30:
By accident I released a few versions of the IRremote library with the IRremoteExtensionTest example containing a IRremoteExtensionTest.cpp instead of a IRremoteExtensionTest.ino file.
It would be nice to add such a check, or do you have another tool which can do this task.

Is the keywords.txt also checked by arduino-lint?

per1234, 2021-05-01:
>It would be nice to add such a check

Thanks for your suggestion!

>Is the keywords.txt also checked by arduino-lint?

Not at this time and I don't know what are the plans specifically for keywords.txt support.

For now, I can recommend my own bash script for checking keywords.txt. There is a workflow here that can simply be copied in to any library repository without modification:
https://github.com/107-systems/107-Arduino-NMEA-Parser/blob/master/.github/workflows/check-keywords-txt.yml

ArminJo, 2022-10-12:
Can I close the issue for me?

per1234, 2022-10-12:
Sure! PRs are always welcome.

closed 2022-10-12 by ArminJo