arduino/arduino-lint #178: Question: Why are library names limited to 16 characters?  (low prio)

RobTillaart, 2021-05-27:
Why are library names limited to 16 characters?
Is there a technical reason?

Rule LP010 result: fail
ERROR: library.properties name value **FunctionGenerator** is longer than the recommended length of 16 characters.

Think the error message should not use the word "recommended" but "maximum" 
"recommended" sounds more like warning level to me 

It is also not mentioned here - https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

Same problem occurred recently with the **DallasTemperatureControlLibrary**

per1234, 2021-05-28:
Hi @RobTillaart. I'm very happy to see you making some use of Arduino Lint!

I have also ran up against this name length limitation when setting up CI for Arduino's libraries with Arduino Lint. In the case of the official libraries, we're even more limited because there is also a requirement that they use the "Arduino_" prefix when Arduino Lint is in the "strict" compliance setting, leaving us with only 8 useful characters.

I have proposed that we consider increasing the limit.

For now, unfortunately the only recourse when using Arduino Lint on libraries with established names longer than 16 characters is to reduce the compliance setting from "strict" to "specification":
https://github.com/arduino/arduino-lint-action#compliance

>Why are library names limited to 16 characters?
Is there a technical reason?

I can think of two problems that might be posed by excessively long library names:
- Layout difficulties when displaying these names
- Difficulty for users who must type the names (e.g., [`arduino-cli lib install`](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_lib_install/))

So I think it's reasonable to have a recommended maximum, but experience is showing that it probably it needs to be a bit longer than 16.

>Think the error message should not use the word "recommended" but "maximum"
>"recommended" sounds more like warning level to me
>
>It is also not mentioned here - https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

This is normally a warning. However, when Arduino Lint's compliance setting is set to "strict", some warnings are upgraded to errors. The idea is to offer the option for those who want to go beyond specification compliance to achieve the gold standard of best practices for Arduino projects. So strict mode is uptight about things the Arduino development software doesn't care about at all, but that we believe provide an improved experience for the users.

The Arduino development software does not impose any maximum name length, and this is why you don't find mention of it in the Arduino Library Specification. So it is correct to say that it's "recommended". Violations of rule LP010 do not result in an error when Arduino Lint is in the "specification" compliance setting that enforces compliance with the official Arduino project specifications.

RobTillaart, 2021-05-28:
Thanks for this extensive explanation!

closed 2021-05-28 by RobTillaart