arduino/arduino-lint #325: Strict check on libraries should check for header guards or #pragma once

Strooom, 2022-01-17:
I think a header file for a library should always have either header guards or #pragme once, so the linter should check for that.
I checked a library without that, and it passed the checks..

per1234, 2022-01-17:
Thanks for your suggestion @Strooom.

Linting of C++ is outside the scope of the Arduino Lint project. The reason is that there are already a tremendous number of excellent tools for C++ code. So there is no need for Arduino to produce another one.

The opportunity for us to provide something of value is in the area of Arduino-specific things such as the `library.properties` metadata file, Library Manager compliance, sketch structure, boards platform configuration files, package index data, etc.

closed 2022-01-17 by per1234

Strooom, 2022-01-17:
I understand, but I think the #pragma once check belongs more to the structure of a good library, than to the C++ contents of it. Anyway, thanks for your fast response and thank you for Arduino Lint