arduino/arduino-lint #348: Feature request: Support macOS arm64

colin-pm, 2022-03-08:
It would be nice to release a version for MacOS arm64 to support the new M1 architecture Macs.

I would be happy to contribute and help test/verify if needed! 😃

per1234, 2022-03-08:
Hi @colin-pm. I don't own a Mac M1, but my understanding is that the x86-64 build we provide already should work fine on these machines using [Rosetta 2 mode](https://support.apple.com/en-us/HT211861).

Have you had problems using Arduino Lint sourced from the "macOS 64 bit" download links shown on the "Installation" page?:

https://arduino.github.io/arduino-lint/latest/installation/#latest-release

colin-pm, 2022-03-08:
I ended up building from source, which worked fine for me. I did just download the Mac x86-64 binary and it appears to work as well.

The only issue I see is the install.sh script errors because getFile returns no binary for macOS arm64. So either hosting a binary for the macOS arm64 download path or modifying the install.sh script would provide support for M1 devices in my mind.

per1234, 2022-03-09:
>I ended up building from source, which worked fine for me. I did just download the Mac x86-64 binary and it appears to work as well.

I'm very glad to hear this!

>the install.sh script errors because getFile returns no binary for macOS arm64

Thanks so much for bringing this to our attention. Although the installation script is automatically tested, including on a macOS x86-64 machine, the GitHub Actions system used for the test does not currently have M1 runners, so that is a gap in the test coverage.

The installation script is maintained in a separate repository, so I have opened a dedicated bug report about this problem there:

https://github.com/arduino/tooling-project-assets/issues/207

Once it is resolved upstream, that will be pulled back into this repository as well as the other projects using the script.

Since it seems no action needs to be taken in the code hosted in the Arduino Lint repository, I'll go ahead and close this issue. 

Regards, Per

closed 2022-03-09 by per1234

per1234, 2022-05-02:
Hello again @colin-pm. I just wanted to share an update that the deficiency in the installation script has now been resolved and it should now be possible to use on M1 machines: https://github.com/arduino/arduino-lint/pull/376

per1234, 2022-07-05:
Hello yet again @colin-pm. I have some good news. There are now native macOS ARM 64-bit builds of Arduino Lint for the Apple M1-based machines (https://github.com/arduino/arduino-lint/pull/409).

There hasn't been a production release of Arduino Lint since the build system was updated, but nightly builds are already available. The link is here if you want to give it a try:

https://arduino.github.io/arduino-lint/dev/installation/#nightly-builds