arduino/arduino-lint #360: arduino-lint can't load url from library.properties (http status 503)

atesin, 2022-03-21:
arduino-lint complaints about unavailable url until it is perfectly available

https://gitlab.com/atesin/xxhash-arduino
```
WARNING: Unable to load the library.properties url field: 503 Service Unavailable (Rule LP042)
```
i am just populating it, i am learning to build and publish a library, i am linting my files locally, once they passes all tests i will upload to gitlab repo

- architecture: windows x64

per1234, 2022-03-21:
Hi @atesin. Thanks for your interest in Arduino Lint.

What other behavior would you suggest in this case?

It seems to me that Arduino Lint is working exactly as intended, since the URL indeed will not load. This is only a warning so you are welcome to disregard it so long as it does not indicate a problem at this point in the development of the project.

atesin, 2022-03-21:
NO WAY! ....  when i created the repo on gitlab it created as private!

thanks to your response i got curious and go to open in a private browser window and then i realized...  fixed!, thanks

now my lib passes all the tests.... would be good to add a note somewhere to avoid these mistakes to someone else in the future, thanks

per1234, 2022-03-21:
I'm glad to hear you found the cause of the rule LP042 violation!

I misunderstood the issue and thought that you were aware that the URL was unavailable.

Best wishes for good progress in the development of your library,
Per

closed 2022-03-21 by per1234