arduino/arduino-lint #756: Release a new version of the arduino-lint

Xayton, 2024-08-05:
Some fixes have been done to `arduino-lint`, release a new version of it.
https://github.com/arduino/arduino-lint/issues/751

per1234, 2024-10-16:
I believe it will be necessary to pull in https://github.com/arduino/tooling-project-assets/pull/469 (and adjust the repository secrets to reflect the change to the workflows) before a release can be made.

I see that the nightly build publishing is already failing at the "**Upload release files on Arduino downloads servers**" step:

https://github.com/arduino/arduino-lint/actions/runs/11357212988/job/31589796943#step:5:17

```text
time="2024-10-16T01:34:16Z" level=info msg="Attempting to upload" bucket=*** endpoint= region=us-east-1
time="2024-10-16T01:34:16Z" level=info msg="Uploading file" bucket=*** name=dist/nightly-20241016-checksums.txt target=/arduino-lint/nightly/nightly-20241016-checksums.txt
time="2024-10-[16](https://github.com/arduino/arduino-lint/actions/runs/11357212988/job/31589796943#step:5:17)T01:34:16Z" level=error msg="Could not upload file" bucket=*** error="InvalidAccessKeyId: The AWS Access Key Id you provided does not exist in our records.\n\tstatus code: 403, request id: A65GYVE37E9YMYYY, host id: OgxUt1Kv4V884Alqfmf8+troHhvYoV8uMJXwCNTuVMPT/IssnVGYSsiu9JNPTHQv2Hehu5ABT68=" name=dist/nightly-20241016-checksums.txt target=/arduino-lint/nightly/nightly-20241016-checksums.txt
time="2024-10-16T01:34:16Z" level=fatal msg="InvalidAccessKeyId: The AWS Access Key Id you provided does not exist in our records.\n\tstatus code: 403, request id: A65GYVE37E9YMYYY, host id: OgxUt1Kv4V884Alqfmf8+troHhvYoV8uMJXwCNTuVMPT/IssnVGYSsiu9JNPTHQv2Hehu5ABT68="
```

and I'm sure the same error would occur if the "Release" workflow was triggered by a release without the workflow having been updated per https://github.com/arduino/tooling-project-assets/pull/469.

per1234, 2024-10-17:
We also need to pull in https://github.com/arduino/arduino-cli/pull/2732 before making a release. If that isn't done then the macOS release won't be notarized due to the regression introduced by pulling in https://github.com/arduino/tooling-project-assets/pull/507 via https://github.com/arduino/arduino-lint/pull/662.

MatteoPologruto, 2024-12-19:
Released here: https://github.com/arduino/arduino-lint/releases/tag/1.3.0

closed 2024-12-19 by MatteoPologruto